epsilonrt/piduino #5: std system error

capriciousduck, 2018-11-28:
When executing pido or pinfo I get the following error

terminate called after throwing an instance of 'std::system_error'
  what():  It seems that this system is not supported !: Operation not supported
Aborted

epsilonrt, 2018-11-28:
Hi,
This exception indicates that the automatic detection did not find the Pi board model.
Could you tell me the Pi board model you use?

capriciousduck, 2018-11-28:
I use the Rpi 3 Model B

epsilonrt, 2018-11-28:
Ok,
can you give me the result of `cat /proc /cpuinfo`?

capriciousduck, 2018-11-28:
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 2a02082
Serial          : 000000004192b6ac

capriciousduck, 2018-11-28:
Thank you very much for your fast reply.

epsilonrt, 2018-11-28:
There is a problem with revision number 0x2a02082, this number is not registered on the site [RPi Hardware](https://elinux.org/RPi_HardwareHistory#Board_Revision_History). Could you give me the link to the page of the builder of this board ?

capriciousduck, 2018-11-28:
I dont know of any page of the builder. I purchased the Pi on amazon an year ago.

capriciousduck, 2018-11-28:
Any I have seen that my board is out of warranty when I installed wiringPi

capriciousduck, 2018-11-28:
Here is the output of gpio -v command 

gpio version: 2.46
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 3, Revision: 02, Memory: 1024MB, Maker: Sony [Out of Warranty]
  * Device tree is enabled.
  *--> Raspberry Pi 3 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.

epsilonrt, 2018-11-28:
How did you install? from the sources?

capriciousduck, 2018-11-28:
No. From the raspbian repository

capriciousduck, 2018-11-28:
via apt-get

epsilonrt, 2018-11-28:
Okay, there is an error in the revision number on the RPi Hardware website (I can not test all the boards versions !!). I would correct in the next version of the database.
For the moment, I propose to modify the file `/etc/piduino.conf`.
Remove the # from the last line to have `revision=0xa02082`

capriciousduck, 2018-11-28:
Ok

capriciousduck, 2018-11-28:
Thank you for your help.

capriciousduck, 2018-11-28:
And I want to control the GPIO via bluetooth using an android app. Should I need to install any other libraries or can it be done via Serial.read()  ?

capriciousduck, 2018-11-28:
I previously used an app to control arduino via bluetooth serial communication. Can I use the same app to control the Pi serial ?

epsilonrt, 2018-11-28:
piduino is designed so that the native arduino code works, so no problem to use the Serial object. no need for additional libraries.

epsilonrt, 2018-11-28:
look at the example [ASCIITable](https://github.com/epsilonrt/piduino/blob/master/examples/communication/ASCIITable/ASCIITable.cpp)

capriciousduck, 2018-11-28:
Hi, I'm still getting the std error

epsilonrt, 2018-11-28:
it should be installed in `/usr/share/doc/piduino/example`

capriciousduck, 2018-11-28:
I've set my revision to  2a02082 without the 0x at the start and executing pinfo, I get that the board is Raspberry Pi B instead of Raspberry Pi 3 B and pido readall showed less than 40 GPIO  pins

epsilonrt, 2018-11-28:
did you modify the configuration file?

capriciousduck, 2018-11-28:
Yes. I just did what you said.

capriciousduck, 2018-11-28:
I remove the # and rebooted the Pi and checked but no success

capriciousduck, 2018-11-28:
Now I set the revision to 2a02082 and executed pinfo I got the info below

Name            : RaspberryPi B
Family          : RaspberryPi
Database Id     : 0
Manufacturer    : Unknown
Board Revision  : 0x2
SoC             : Bcm2708 (Broadcom)
Memory          : 1024MB
GPIO Id         : 1
PCB Revision    : 1
Serial Ports    : /dev/ttyAMA0

See it is showing as Raspi B instead of RPi 3B

capriciousduck, 2018-11-28:
pido readall gives me this 

      P1 (#1)
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
| sOc | iNo |   Name   | Mode | V | Ph || Ph | V | Mode |   Name   | iNo | sOc |
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
|     |     |     3.3V |      |   |  1 || 2  |   |      | 5V       |     |     |
|   0 |  30 |     SDA0 | ALT0 | 0 |  3 || 4  |   |      | 5V       |     |     |
|   1 |  31 |     SCL0 | ALT0 | 0 |  5 || 6  |   |      | GND      |     |     |
|   4 |   7 |    GPIO4 | ALT4 | 1 |  7 || 8  | 0 | ALT4 | GPIO14   | 15  | 14  |
|     |     |      GND |      |   |  9 || 10 | 1 | ALT4 | GPIO15   | 16  | 15  |
|  17 |   0 |     RTS1 | ALT5 | 1 | 11 || 12 | 1 | ALT4 | SPI1CE0  | 1   | 18  |
|  21 |  29 |   GPIO21 |   IN | 1 | 13 || 14 |   |      | GND      |     |     |
|  22 |   3 |  ARMTRST | ALT4 | 1 | 15 || 16 | 0 | OUT  | GPIO23   | 4   | 23  |
|     |     |     3.3V |      |   | 17 || 18 | 1 | ALT4 | ARMTDO   | 5   | 24  |
|  10 |  12 | SPI0MOSI | ALT0 | 1 | 19 || 20 |   |      | GND      |     |     |
|   9 |  13 |    GPIO9 | ALT2 | 1 | 21 || 22 | 1 | ALT4 | ARMTCK   | 6   | 25  |
|  11 |  14 | SPI0SCLK | ALT0 | 1 | 23 || 24 | 1 | ALT4 | GPIO8    | 10  | 8   |
|     |     |      GND |      |   | 25 || 26 | 0 | ALT5 | GPIO7    | 11  | 7   |
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
| sOc | iNo |   Name   | Mode | V | Ph || Ph | V | Mode |   Name   | iNo | sOc |
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+

capriciousduck, 2018-11-28:
Only 26 pins instead of 40 GPIO

epsilonrt, 2018-11-28:
No no ! 0x2a02082 is not in the database ! you have to leave `0xa02082`, that corresponds to your board:
`27|RaspberryPi 3 (0xA02082)|1024|1.2|7|3|1|1|0|0`

capriciousduck, 2018-11-28:
So what is the revision I should put in the file ? Can you paste it here ?

epsilonrt, 2018-11-28:
you must have `revision=0xa02082` on the last line

capriciousduck, 2018-11-28:
Yeah I changed it to the above revision, but still I get the same error

capriciousduck, 2018-11-28:
here is my piduino.conf file

cat /etc/piduino.conf
# PiDuino configuration file
connection_info="sqlite3:db=/usr/share/piduino/piduino.db"

# Allows you to force the board tag (Armbian)
# !! Be careful, this is at your own risk !!
# !! Forcing an incorrect value may destroy GPIO pins !!
#tag="nanopineocore"

# Allows forcing the revision of the board (Raspbian)
# !! Be careful, this is at your own risk !!
# !! Forcing an incorrect value may destroy GPIO pins !!
revision=0xa02082

epsilonrt, 2018-11-28:
```
# Allows forcing the revision of the board (Raspbian)
# !! Be careful, this is at your own risk !!
# !! Forcing an incorrect value may destroy GPIO pins !!
revision=0xa02082
```

epsilonrt, 2018-11-28:
great, that's it

capriciousduck, 2018-11-28:
But still I get the error. It's not working

epsilonrt, 2018-11-28:
it works ?

capriciousduck, 2018-11-28:
no.

capriciousduck, 2018-11-28:
I still get the std system error

epsilonrt, 2018-11-28:
argh !

capriciousduck, 2018-11-28:
what should I do now ?

epsilonrt, 2018-11-28:
can you send me the file /etc/piduino.conf by mail on epsilonrt@gmail.com?

capriciousduck, 2018-11-28:
Yop, will send it right now.

capriciousduck, 2018-11-28:
Check your inbox. I've sent the file

epsilonrt, 2018-11-28:
I received ! the file is good, it should work ...
I found the page with the correct revision numbers:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md

I will update the database and update the package, but it will not be until this afternoon because there I have no raspberry pi available ...

capriciousduck, 2018-11-28:
You mean you will update the package by the end of day ?

epsilonrt, 2018-11-28:
Let me know if it works ... I warn you when the package has been updated.

capriciousduck, 2018-11-28:
Sure !

capriciousduck, 2018-11-28:
are all arduino commands supported in piduino ?

epsilonrt, 2018-11-28:
Yes !
that's the purpose of the thing!

capriciousduck, 2018-11-28:
so if i connect my android app to raspberry pi via bluetooth and with just Serial.read() can i receive data to raspberry pi from my android app ?

capriciousduck, 2018-11-28:
I would like to discuss about this in detail after the revision issue is fixed.

epsilonrt, 2018-11-28:
I just analyzed the problem, I'll have to modify the code because your revision number has its F bit to 1. On older version of Raspbian, this bit was hidden. So it will take a little longer ...

capriciousduck, 2018-11-28:
How many days ?

capriciousduck, 2018-11-28:
I was very eager to use piduino in my project ? Is there any temporary fix so that I can use Piduino until you update the code ?

epsilonrt, 2018-11-28:
could you give me the version of raspbian that you use (the date), thanks

capriciousduck, 2018-11-28:
Yup, the latest one. I just downloaded it today.

epsilonrt, 2018-11-28:
Ok, I'm currently going with my students ... it's not easy. This will be done in the evening, here it is 16:00 at the latest tomorrow morning.
Normally, manual modification of the piduino.conf file should have solved the problem, so there is a bug. I need to recreate your environment to understand why it does not work. piduino is in the development phase, even if it is already very stable, my students will use it in their project of end of study, it is perhaps your case?

commit bfc4d465 (2018-11-29): Fixes #5
closed 2018-11-29 by epsilonrt