epsilonrt/piduino #9: rfcomm error

capriciousduck, 2018-11-29:
Hi, I created a sample bluetooth controlled led using piduino library. Below is the code is executed. How can I include rfcomm in the below code ? I am able to run the code, but not able to control the LED using android via bluetooth.

#include <Piduino.h>

int led = 21;
int data = 0;

void setup()

{

  pinMode(led, OUTPUT);
  Serial.begin(9600);

}

void loop()
{

 data = Serial.read();
 if(data == '1')
 {
  digitalWrite(led, HIGH);
 }

 if(data == '0')
 {
  digitalWrite(led, LOW);
 }

}

epsilonrt, 2018-11-29:
I think that a test is missing if Serial.available (), but it should still work because the read () function is non-blocking and returns -1 if nothing is available.
Are you sure you have configured the serial port correctly (by default it is used to display kernel messages):
https://www.raspberrypi.org/documentation/configuration/uart.md
The first thing to do is to watch with minicom if you receive the serial port information ...
With Codelite you can put a breakpoint and see what your program does.
Could you copy me the result of pinfo?

capriciousduck, 2018-11-30:
Here is my pinfo result

Name            : RaspberryPi 3B
Family          : RaspberryPi
Database Id     : 27
Manufacturer    : Sony
Board Revision  : 0xa02082
SoC             : Bcm2710 (Broadcom)
Memory          : 1024MB
GPIO Id         : 3
PCB Revision    : 1.2
Serial Ports    : /dev/ttyAMA0

capriciousduck, 2018-11-30:
Yes, I removed the console=serial0,115200 from my cmdline.txt file but still I was not able to send data to my pi with android.

instead of Serial.begin() can i use the rfcomm in the code ? If yes, how ?

capriciousduck, 2018-11-30:
I want to use RPi 3 bluetooth to receive data instead of using a HC-05. And from what I know is, Serial.read() is used to receive data from a serial bluetooth device like HC-05. How can I use my rfcomm port to receive data using Piduino library ?

epsilonrt, 2018-11-30:
Okay,
Do characters appear in minicom?
`minicom -b9600 -D /dev/ttyAMA0`

capriciousduck, 2018-11-30:
I haven't checked minicom, but I connect my android to rfcomm and I executed cat /dev/rfcomm0, I saw characters

capriciousduck, 2018-11-30:
How can I open rfcomm in arduino code instead of Serial.read() ?

epsilonrt, 2018-11-30:
Can you confirm that the characters are coming to /dev/ttyAMA0 ?

epsilonrt, 2018-11-30:
As long as you do not confirm that the characters are displayed in minicom on the raspberry pi, we can not progress ...

capriciousduck, 2018-11-30:
Should I change any commands in cmdline.txt and config.txt to make minicom work properly ?

capriciousduck, 2018-11-30:
Do  I need any serial adapter or can I use my HC-05 module ?

epsilonrt, 2018-11-30:
Could you explain to me clearly what are the raspberry pi UART0_TXD and UART0_RXD pins connected to?

epsilonrt, 2018-11-30:
That's the result of 
[image]
[ASCIIPing](https://github.com/epsilonrt/piduino/blob/master/examples/communication/ASCIIPing/ASCIIPing.cpp) on a Raspberry Pi 3 (same model as yours). Piduino works very well.
[image]

epsilonrt, 2018-11-30:
Sur le raspberry pi 3, UART0 est utilisé par défaut pour le bluetooth pour le désactiver il faut avoir ça dans /boot/config.txt : 

> #enable_uart=1
> dtoverlay=pi3-disable-bt

capriciousduck, 2018-11-30:
No, I want to communicate via raspberry pi bluetooth

capriciousduck, 2018-11-30:
Okay, sorry I'll make it clear. I want to use raspberry pi bluetooth. How can I use serial.read() with on-board raspberry pi bluetooth ?

epsilonrt, 2018-11-30:
Ok, I understand now.
I have never used bluetooth on a Raspberry Pi 3.
What is the name of the /dev file ?

Currently pinfo only displays /dev/ttyAMA0 which has nothing to do with your bluetooth link, it is normal that you do not receive anything ! Piduino uses libudev to detect serial ports and will create the necessary SerialX objects. I did not consider this requirement in my development.

epsilonrt, 2018-11-30:
I do not have time right now to develop new features (very specific to Raspberry Pi 3), but I would probably do it.

What is certain is that as long as you will not be able to display in minicom the characters received on the bluetooth serial link, you will not be able to make a program waiting for the characters on the serial port ...

capriciousduck, 2018-11-30:
Okay, so then how can I use HC-05 via piduino ?

epsilonrt, 2018-11-30:
I do not know this device, do you have a link?

capriciousduck, 2018-11-30:
here it is. a tutu on it.

https://howtomechatronics.com/tutorials/arduino/arduino-and-hc-05-bluetooth-module-tutorial/

epsilonrt, 2018-11-30:
Ok, seen.
Yes, of course.
You need to turn off the bluetooth in /boot/config.txt as above.

capriciousduck, 2018-11-30:
Okay. Can i call python code in piduino code ?

epsilonrt, 2018-11-30:
No, Arduino is C++ ! that goes against the goal pursued by Piduino.

epsilonrt, 2018-11-30:
the example of your tutorial should work perfectly by adding #include <Piduino.h> and assigning the correct pin to the led. bye.

capriciousduck, 2018-11-30:
Okay. Thank you Professor!

epsilonrt, 2018-12-01:
http://www.raspberry-projects.com/pi/pi-operating-systems/raspbian/bluetooth/serial-over-bluetooth