epsilonrt/piduino #20: It seems that this system is not supported !: Operation not supported

Trisse123, 2020-01-05:
Hi!

Atfirst, thanks for you great work.

I have to communicate with 3 sensors from Sparkfun.
They provide nice libs for the sensors but only for Arduino, so i am searching for exact some kind of library like yours. Sadly, i am not entirely able to port the libs myself to c++ code that can be used in linux, so your lib comes in handy.

Sadly, i got an issue while with my RPI 4.
I am able to complie your lib within my code, but executing the resulting program outputs:

> terminate called after throwing an instance of 'std::system_error'
>   what():  It seems that this system is not supported !: Operation not supported
> Abgebrochen

When i run "pinfo", its the same.

I tried to force the board-revision via /etc/piduino.conf, but i seems that piduino does not even care.
(same error)

I than tried to add my RPI4 to the /usr/share/piduino/piduino.db
What i added:
Table board: 45 | RaspberryPi 4 B | 4096 | 1.1 | 3 | 3 | 1 | 1 | 0 | 0
Table board_model: 22 | RaspberryPi 4 B | 0 | 5
Table revision: 45 | 10821763
Table soc: 5 | BCM2835 | 0 | 0 | 2 | 1 | 1

But again, piduino gives me the "not supported" message.
I dont know, if i have entered alle the values correct, but i felt like a fill-in-the-blank text, so it cant be that wrong.
(See attached piduino.db, if you want
[piduino.zip](https://github.com/epsilonrt/piduino/files/4023800/piduino.zip)

)

I than compiled piduino myself from source: same error.

It seems forcing the revision is not possible and adding the RPI manually to the DB also not.
Am i doing somesting wrong?

I am using latest Raspbian Buster image.

**I would be so glad if you would try to help me here.**

At last, here is an output of /proc/cpuinfo, if you need it.

> processor       : 0
> model name      : ARMv7 Processor rev 3 (v7l)
> BogoMIPS        : 108.00
> Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
> CPU implementer : 0x41
> CPU architecture: 7
> CPU variant     : 0x0
> CPU part        : 0xd08
> CPU revision    : 3
> 
> processor       : 1
[piduino.zip](https://github.com/epsilonrt/piduino/files/4023801/piduino.zip)

> model name      : ARMv7 Processor rev 3 (v7l)
> BogoMIPS        : 108.00
> Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
> CPU implementer : 0x41
> CPU architecture: 7
> CPU variant     : 0x0
> CPU part        : 0xd08
> CPU revision    : 3
> 
> processor       : 2
> model name      : ARMv7 Processor rev 3 (v7l)
> BogoMIPS        : 108.00
> Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
> CPU implementer : 0x41
> CPU architecture: 7
> CPU variant     : 0x0
> CPU part        : 0xd08
> CPU revision    : 3
> 
> processor       : 3
> model name      : ARMv7 Processor rev 3 (v7l)
> BogoMIPS        : 108.00
> Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
> CPU implementer : 0x41
> CPU architecture: 7
> CPU variant     : 0x0
> CPU part        : 0xd08
> CPU revision    : 3
> 
> Hardware        : BCM2835
> Revision        : a03111
> Serial          : 100000002da3ffc0
> Model           : Raspberry Pi 4 Model B Rev 1.1

Trisse123, 2020-01-05:
Ok i now got something else:
I purged the libs again and rebuilded again from source.
I again checked the added values in the DB and i saw, that there is was a typo in the revision-number.

I fixed this, and now piduino gives me the following:

> terminate called after throwing an instance of 'std::out_of_range'
>   what():  map::at
> Abgebrochen

Could you please try to help me?

epsilonrt, 2020-01-05:
Hi,
I see that you worked well before asking me the question.
I have plans to extend pinfo to allow it to add boards to the database, but I have a lot of work at the moment ... This should be done around May.
I don't know RPI4, I'll think it over and give you an answer tonight.

epsilonrt, 2020-01-05:
I do not see a difference between the GPIO of the Raspberry Pi 3 and that of the 4, moreover, the processor seems to me equivalent.
Piduino uses the board revision number to identify the Raspberry Pi.
You could try to modify /etc/piduino.conf by putting the revision number of a Raspberry Pi 3, for example:
Revision = 0xa32082
piduino will behave as if you had an RPI3.
I advise you to reinstall the deb version before doing the test.
Keep me informed.

epsilonrt, 2020-01-05:
Tu peux utiliser la version deb en installant le dépôt buster

```
wget -O- http://www.piduino.org/piduino-key.asc | sudo apt-key add -
echo 'deb http://raspbian.piduino.org buster piduino' | sudo tee /etc/apt/sources.list.d/piduino.list
sudo apt update
sudo apt install libpiduino-dev piduino-utils
```

epsilonrt, 2020-01-05:
All revisions are [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md)
Since I have not worked with the Raspberry Pi for a while, the last revision in the database is a32082.

Trisse123, 2020-01-07:
Thanks for that - I see you are willin to help me.
But as I said i already tried to force a revision in /etc/piduino.conf.
I also tried "revision=0xa32082", but again "It seems that this system is not supported".

I also have no luck with that sqlite db.
I tried many values and double-checked everything, but no luck..

_Side-note: when i install piduino via debs (apt install libpiduino-dev piduino-utils), like in my first time, /etc/piduino.conf does not beeing created.
When i purge everyting (apt purge libpiduino-dev piduino-utils) and build piduino from source, /etc/piduino.conf gets created.
-> Only for you to know - maybe a bug_

Yes, you are right, RPI3 and 4-Versions are roughly the same, so piduino sould have no problem with that, but sadly it does.
Could you try to update piduino according to my (RPI4's) values?
Do you need any other values from me?
Please assist me, if you want me to test something!
I would like to contribute to make piduino even better - but at first i need to get it to work :P
(i even could set up a remote ssh access to my RPI4, if like to try and test some things)

epsilonrt, 2020-01-08:
Bonjour,
I did not see in your initial report that you had try the value 0xa32082 ...
Currently, the only way to understand the origin of exceptions is to compile with `cmake -DCMAKE_BUILD_TYPE = Debug ..` and see the faulty line with a debugger (codelite, gdb, cgdb ...). But I don't have an RPI4 ... :-(

The exception "It seems that this system is not supported!" comes from line 55 of the file [gpio.cpp] (https://github.com/epsilonrt/piduino/blob/master/src/gpio/gpio.cpp). This indicates that the board model has not been detected, which is normal for the RPI4. BUT it is not normal when you force the revision number to 0xa32082.

You MUST ABSOLUTELY uninstall the version you have built:

cd cmake-build-Release
sudo make uninstall
sudo ldconfig

Check that the files `/etc/piduino.conf` and `/usr/local/share/piduino/piduino.db` are deleted (backup copy, if you want to keep them).

Then you install from deb packages and modify `/etc/piduino.conf` with `revision=0xa32082` (remove the # at the beginning of the line !!)

It MUST work.

As for your modification of the database, adding a new SOC (BCM2711) can only be done by modifying the source code. It is my choice to avoid using a code that is not made for a SOC that is not planned. It seems logical to me.

The exception below is related to line 28 of the file [bcm2835.cpp](https://github.com/epsilonrt/piduino/blob/master/src/arch/arm/broadcom/bcm2835.cpp)

> terminate called after throwing an instance of 'std::out_of_range'
> what(): map::at

For now I don't have time to think about adding a new SOC, but normally you should be able to make piduino work by passing your RPI4 for an RPI3. this is the path you should follow.

epsilonrt, 2020-01-09:
Hi,
It seems that the base address of the GPIO has changed on the RPI4.
Could you give me the result of the command `cat /proc/iomem | grep gpio@` ?
Thanks

epsilonrt, 2020-01-09:
The iomem base address has changed, so you can't fool piduino. I ordered an RPI4 on Amazon, reception tomorrow, you should have an update with support for RPI4 over the weekend.

Trisse123, 2020-01-09:
You are awesome! 🥇 

Thanks so much for fiddeling with that issue and of course for taking your time and money to buy you an RPI4. I dont think that many guys would do that for free..

Here is my whole `cat /proc/iomem`, just for the sake of completeness 

> 00000000-3b3fffff : System RAM
>   00008000-00dfffff : Kernel code
>   01000000-0117a04f : Kernel data
> fd500000-fd50930f : pcie@7d500000
> fd580000-fd58ffff : genet@7d580000
>   fd580e14-fd580e1c : unimac-mdio.-19
> fd5d2200-fd5d222b : thermal@7d5d2200
> fe007000-fe007aff : dma@7e007000
> fe007b00-fe007eff : dma@7e007b00
> fe00a000-fe00a023 : watchdog@7e100000
> fe00b200-fe00b3ff : dwc_otg
> fe00b840-fe00b87b : mailbox@7e00b840
> fe00b880-fe00b8bf : mailbox@7e00b880
> fe100000-fe100113 : watchdog@7e100000
> fe101000-fe102fff : cprman@7e101000
> fe104000-fe10400f : rng@7e104000
> fe200000-fe2000b3 : gpio@7e200000
> fe201000-fe2011ff : serial@7e201000
>   fe201000-fe2011ff : serial@7e201000
> fe215000-fe215007 : aux@7e215000
> fe300000-fe3000ff : mmcnr@7e300000
> fe340000-fe3400ff : emmc2@7e340000
> fe600000-fe6000ff : firmwarekms@7e600000
> fe804000-fe804fff : i2c@7e804000
> fe980000-fe98ffff : dwc_otg
> fec00000-fec03fff : hub
> fec04000-fec07fff : core0
> fec11000-fec1101f : watchdog@7e100000
> 600000000-603ffffff : pcie@7d500000
>   600000000-6000fffff : PCI Bus 0000:01
>     600000000-600000fff : 0000:01:00.0
>       600000000-600000fff : xhci-hcd

Again, thank you so much.
For sure some other guys will benefint from adding RPI4 support in the future.
Let me buy you a beer when this issue is resolved.
But.. you are french, he? Better let me buy you a good wine and some camembert 👍 

**merci d'avance**,
salutations de l'allemagne

epsilonrt, 2020-01-10:
The PWM functionality and the control of the pull resistors are not yet operational.
I'm working on it, I'll let you know when you can use it.

epsilonrt, 2020-01-11:
the work is finished, i have carried out the tests and everything works.
for now, everything is in the git r0.6 branch but I pushed the packages into my raspbian repository.
Here are the manipulations to do:
Uninstall any version installed from sources !! IMPORTANT
then reboot.

```
wget -O- http://www.piduino.org/piduino-key.asc | sudo apt-key add -
echo 'deb http://raspbian.piduino.org stretch piduino' | sudo tee /etc/apt/sources.list.d/piduino.list
sudo apt update
sudo apt install libpiduino-dev piduino-utils

pascal@raspberrypi:~$ pinfo --version
Piduino pinfo version 0.5-14
Copyright © 2018-2019 Pascal JEAN, https://github.com/epsilonrt/piduino
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'pinfo -w' for details.

pascal@raspberrypi:~$ pinfo
Name            : RaspberryPi 4B
Family          : RaspberryPi
Database Id     : 50
Manufacturer    : Sony UK
Board Revision  : 0xa03111
SoC             : Bcm2711 (Broadcom)
Memory          : 1024MB
GPIO Id         : 3
PCB Revision    : 1.1
I2C Buses       : /dev/i2c-1
SPI Buses       : /dev/spidev0.0,/dev/spidev0.1
Serial Ports    : /dev/ttyAMA0,/dev/ttyS0
pascal@raspberrypi:~$ pido readall
                                           J8 (#1)
+-----+-----+----------+------+------+---+----++----+---+------+------+----------+-----+-----+
| sOc | iNo |   Name   | Mode | Pull | V | Ph || Ph | V | Pull | Mode |   Name   | iNo | sOc |
+-----+-----+----------+------+------+---+----++----+---+------+------+----------+-----+-----+
|     |     |     3.3V |      |      |   |  1 || 2  |   |      |      | 5V       |     |     |
|   2 |   8 |     SDA1 | ALT0 |   UP | 1 |  3 || 4  |   |      |      | 5V       |     |     |
|   3 |   9 |     SCL1 | ALT0 |   UP | 1 |  5 || 6  |   |      |      | GND      |     |     |
|   4 |   7 |    GPIO4 |   IN |   UP | 0 |  7 || 8  | 1 | OFF  | ALT5 | TXD1     | 15  | 14  |
|     |     |      GND |      |      |   |  9 || 10 | 1 | UP   | ALT5 | RXD1     | 16  | 15  |
|  17 |   0 |   GPIO17 |   IN | DOWN | 1 | 11 || 12 | 1 | DOWN | IN   | GPIO18   | 1   | 18  |
|  27 |   2 |   GPIO27 |   IN | DOWN | 1 | 13 || 14 |   |      |      | GND      |     |     |
|  22 |   3 |   GPIO22 |   IN | DOWN | 0 | 15 || 16 | 0 | DOWN | IN   | GPIO23   | 4   | 23  |
|     |     |     3.3V |      |      |   | 17 || 18 | 0 | DOWN | IN   | GPIO24   | 5   | 24  |
|  10 |  12 | SPI0MOSI | ALT0 | DOWN | 0 | 19 || 20 |   |      |      | GND      |     |     |
|   9 |  13 | SPI0MISO | ALT0 | DOWN | 0 | 21 || 22 | 0 | DOWN | IN   | GPIO25   | 6   | 25  |
|  11 |  14 | SPI0SCLK | ALT0 | DOWN | 0 | 23 || 24 | 1 | UP   | OUT  | GPIO8    | 10  | 8   |
|     |     |      GND |      |      |   | 25 || 26 | 1 | UP   | OUT  | GPIO7    | 11  | 7   |
|   0 |  30 |    GPIO0 |   IN |   UP | 1 | 27 || 28 | 1 | UP   | IN   | GPIO1    | 31  | 1   |
|   5 |  21 |    GPIO5 |   IN |   UP | 1 | 29 || 30 |   |      |      | GND      |     |     |
|   6 |  22 |    GPIO6 |   IN |   UP | 1 | 31 || 32 | 0 | DOWN | IN   | GPIO12   | 26  | 12  |
|  13 |  23 |   GPIO13 |   IN | DOWN | 0 | 33 || 34 |   |      |      | GND      |     |     |
|  19 |  24 |   GPIO19 |   IN | DOWN | 0 | 35 || 36 | 0 | DOWN | IN   | GPIO16   | 27  | 16  |
|  26 |  25 |   GPIO26 |   IN | DOWN | 0 | 37 || 38 | 0 | DOWN | IN   | GPIO20   | 28  | 20  |
|     |     |      GND |      |      |   | 39 || 40 | 0 | DOWN | IN   | GPIO21   | 29  | 21  |
+-----+-----+----------+------+------+---+----++----+---+------+------+----------+-----+-----+
| sOc | iNo |   Name   | Mode | Pull | V | Ph || Ph | V | Pull | Mode |   Name   | iNo | sOc |
+-----+-----+----------+------+------+---+----++----+---+------+------+----------+-----+-----+

```

Note that on the BCM2711 you can read the state of the pull resistors !
Put a led on pin 0 (GPIO17) and pin 1 (GPIO18) and a push button on pin 3 (GPIO22) then :

```

pascal@raspberrypi:~$ pido blink 0
Press Ctrl+C to abort ...
^C
everything was closed.
Have a nice day !
pascal@raspberrypi:~$ pido mode 3 in
pascal@raspberrypi:~$ pido pull 3 up
pascal@raspberrypi:~$ pido wfi 3 falling
0
pascal@raspberrypi:~$ pido mode 1 pwm
pascal@raspberrypi:~$ pido pwm 1 1023
pascal@raspberrypi:~$ pido pwm 1 512

pascal@raspberrypi:~$ cp -a /usr/share/doc/piduino/examples/Blink .
pascal@raspberrypi:~$ cd Blink/
pascal@raspberrypi:~/Blink$ g++ -o Blink Blink.cpp $(pkg-config --cflags --libs piduino)
pascal@raspberrypi:~/Blink$ sudo ./Blink 

```
Could you proceed with the installation and the test and bring me back any problems?

closed 2020-01-11 by epsilonrt

Trisse123, 2020-01-14:
Hi!

I was now able to compile my code against your libs.
This is so awesome! Thanks so much for your work adding RPI4 support.
I am so glad you helped me.

👍

epsilonrt, 2020-01-14:
Hi,
I'm happy to see that my work suits you. 
I think you could star piduino on github ?
Can you also tell your colleagues about piduino ?
Piduino is not yet in version 1.0 but should be released at the end of the year.
I would need a little help on translating the documentation because I am not very good at English ... :-(
So if you have a little time available, it is not necessarily urgent ... I would like you to correct my README files and piduino wiki. You can do it very simply by fork piduino then after having updated the documentation, you could make pull requests.

JetForMe, 2021-10-09:
Is this working on pi4 now? I just installed the buster version, and I get the same error:

```
$ pinfo
terminate called after throwing an instance of 'std::system_error'
  what():  It seems that this system is not supported !: Operation not supported
Aborted
```

arvin-tan, 2021-12-22:
> Is this working on pi4 now? I just installed the buster version, and I get the same error:
> 
> ```
> $ pinfo
> terminate called after throwing an instance of 'std::system_error'
>   what():  It seems that this system is not supported !: Operation not supported
> Aborted
> ```

I second this. Followed the instructions above but to no avail.

commit b68f26fb (2020-01-11): Fixes #20, all RPi4 features are functional.

- Cancellation of commit #b70b5d4,
- Change of the PWM clock frequency for the BCM2711 (54MHz instead of 19.2MHz).