epsilonrt/piduino #26: Raspberry pi CM4 support

dolfje, 2023-03-13:
I'm busy with trying piduino for the Raspberry Pi CM4. 

From what I read, it is almost the same as  Raspberry 4b.
So I'm trying to get it working.

I have already updated the database to the following:

board:
54|RaspberryPi Compute Module 4 (0xb03140)|2048|1.0|24|3|1|1|0|0
board_family:
24|RaspberryPi Compute Module 4|0|5
revision:
54|11546944

Now busy with checking if this works. So will update when I find incompatibilities.

dolfje, 2023-03-14:
The CM4 has 5 serial ports, so https://github.com/epsilonrt/piduino/blob/8b791094fb501583dcf9096b4d09cbbbfdc381cc/include/piduino/arduino/HardwareSerial.h#L112 has to be changed.

epsilonrt, 2025-04-09:
Hi,
I apologize for the late reply.
It's not that simple; in the case of CM4, you'd have to create a new GPIO connector...
I'm still planning to create a database client [pidbm](https://github.com/epsilonrt/pidbm) to do this easily, but I'm running out of time.

epsilonrt, 2025-04-11:
Support for your board was added in commit [9a414b4](https://github.com/epsilonrt/piduino/commit/9a414b46f530c4b523eeeef5308527aabd805bad).
If you have the opportunity, please confirm that everything is working.

closed 2025-04-11 by epsilonrt

epsilonrt, 2025-04-12:
```
epsilonrt@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
epsilonrt@raspberrypi:~ $ pido -v
Piduino pido version 0.5-58
Copyright © 2018-2019 Pascal JEAN, https://github.com/epsilonrt/piduino
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'pido -w' for details.

epsilonrt@raspberrypi:~ $ pinfo
Name            : Raspberry Pi Compute Module 4
Model           : RaspberryPi Compute Module 4
Family          : RaspberryPi
Database Id     : 68
Manufacturer    : Sony UK
Board Revision  : 0xa03141
SoC             : Bcm2711 (Broadcom)
Memory          : 1024MB
GPIO Id         : 3
PCB Revision    : 1.1
I2C Buses       : /dev/i2c-20,/dev/i2c-21
epsilonrt@raspberrypi:~ $ pido readall
                                           J8 (#1)
+-----+-----+----------+------+------+---+----++----+---+------+------+----------+-----+-----+
| sOc | iNo |   Name   | Mode | Pull | V | Ph || Ph | V | Pull | Mode |   Name   | iNo | sOc |
+-----+-----+----------+------+------+---+----++----+---+------+------+----------+-----+-----+
|     |     |     3.3V |      |      |   |  1 || 2  |   |      |      | 5V       |     |     |
|   2 |   8 |    GPIO2 |   IN |   UP | 1 |  3 || 4  |   |      |      | 5V       |     |     |
|   3 |   9 |    GPIO3 |   IN |   UP | 1 |  5 || 6  |   |      |      | GND      |     |     |
|   4 |   7 |    GPIO4 |   IN |   UP | 1 |  7 || 8  | 1 | OFF  | IN   | GPIO14   | 15  | 14  |
|     |     |      GND |      |      |   |  9 || 10 | 1 | UP   | IN   | GPIO15   | 16  | 15  |
|  17 |   0 |   GPIO17 |   IN | DOWN | 0 | 11 || 12 | 0 | DOWN | IN   | GPIO18   | 1   | 18  |
|  27 |   2 |   GPIO27 |   IN | DOWN | 0 | 13 || 14 |   |      |      | GND      |     |     |
|  22 |   3 |   GPIO22 |   IN | DOWN | 0 | 15 || 16 | 0 | DOWN | IN   | GPIO23   | 4   | 23  |
|     |     |     3.3V |      |      |   | 17 || 18 | 0 | DOWN | IN   | GPIO24   | 5   | 24  |
|  10 |  12 |   GPIO10 |   IN | DOWN | 0 | 19 || 20 |   |      |      | GND      |     |     |
|   9 |  13 |    GPIO9 |   IN | DOWN | 0 | 21 || 22 | 0 | DOWN | IN   | GPIO25   | 6   | 25  |
|  11 |  14 |   GPIO11 |   IN | DOWN | 0 | 23 || 24 | 1 | UP   | IN   | GPIO8    | 10  | 8   |
|     |     |      GND |      |      |   | 25 || 26 | 1 | UP   | IN   | GPIO7    | 11  | 7   |
|   0 |  30 |    GPIO0 |   IN |   UP | 1 | 27 || 28 | 1 | UP   | IN   | GPIO1    | 31  | 1   |
|   5 |  21 |    GPIO5 |   IN |   UP | 1 | 29 || 30 |   |      |      | GND      |     |     |
|   6 |  22 |    GPIO6 |   IN |   UP | 1 | 31 || 32 | 0 | DOWN | IN   | GPIO12   | 26  | 12  |
|  13 |  23 |   GPIO13 |   IN | DOWN | 0 | 33 || 34 |   |      |      | GND      |     |     |
|  19 |  24 |   GPIO19 |   IN | DOWN | 0 | 35 || 36 | 0 | DOWN | IN   | GPIO16   | 27  | 16  |
|  26 |  25 |   GPIO26 |   IN | DOWN | 0 | 37 || 38 | 0 | DOWN | IN   | GPIO20   | 28  | 20  |
|     |     |      GND |      |      |   | 39 || 40 | 0 | DOWN | IN   | GPIO21   | 29  | 21  |
+-----+-----+----------+------+------+---+----++----+---+------+------+----------+-----+-----+
| sOc | iNo |   Name   | Mode | Pull | V | Ph || Ph | V | Pull | Mode |   Name   | iNo | sOc |
+-----+-----+----------+------+------+---+----++----+---+------+------+----------+-----+-----+

epsilonrt@raspberrypi:~ $ pido blink 0
Press Ctrl+C to abort ...
^C
everything was closed.
Have a nice day !
```

The revision number of the board used is not in the database, its information was automatically extracted by piduino by decoding the version number. However, everything works.