epsilonrt/piduino #30: Raspberry pi Zero 2 W support

dolfje, 2023-08-07:
I'm busy with trying piduino for the Raspberry Pi Zero 2 W.

From what I read, it is almost the same as Raspberry 3A.
So I'm trying to get it working.

I have already updated the database to the following:

board:
55|RaspberryPi Zero 2 Wifi (902120)|512|1.0|24|3|1|1|0|0
board_model:
25|RaspberryPi Zero 2 Wifi|0|2
revision:
55|9445664

Now busy with checking if this works. So will update when I find incompatibilities.

epsilonrt, 2025-04-09:
hi,
I've started a long-term project to integrate all Raspberry Pi boards.
However, I don't have a Zero 2W board, so if you're still interested, I'll let you know when the new version will be released so you can confirm it works.

epsilonrt, 2025-04-11:
Support for your board was added in commit [9a414b4](https://github.com/epsilonrt/piduino/commit/9a414b46f530c4b523eeeef5308527aabd805bad).
If you have the opportunity, please confirm that everything is working.

closed 2025-04-11 by epsilonrt

epsilonrt, 2025-04-12:
```
epsilonrt@rpi-bookworm-arm64:~ $ pido -v
Piduino pido version 0.5-52
Copyright © 2018-2019 Pascal JEAN, https://github.com/epsilonrt/piduino
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'pido -w' for details.

epsilonrt@rpi-bookworm-arm64:~ $ pinfo
Name            : RaspberryPi Zero 2 W (0x902120)
Model           : RaspberryPi Zero 2 W
Family          : RaspberryPi
Database Id     : 63
Manufacturer    : Sony UK
Board Revision  : 0x902120
SoC             : Bcm2710 (Broadcom)
Memory          : 512MB
GPIO Id         : 3
PCB Revision    : 1.0
I2C Buses       : /dev/i2c-1,/dev/i2c-2
SPI Buses       : /dev/spidev0.0,/dev/spidev0.1
Serial Ports    : /dev/ttyS0
epsilonrt@rpi-bookworm-arm64:~ $ pido readall
                                    J8 (#1)
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
| sOc | iNo |   Name   | Mode | V | Ph || Ph | V | Mode |   Name   | iNo | sOc |
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
|     |     |     3.3V |      |   |  1 || 2  |   |      | 5V       |     |     |
|   2 |   8 |     SDA1 | ALT0 | 1 |  3 || 4  |   |      | 5V       |     |     |
|   3 |   9 |     SCL1 | ALT0 | 1 |  5 || 6  |   |      | GND      |     |     |
|   4 |   7 |    GPIO4 |   IN | 1 |  7 || 8  | 1 | ALT5 | TXD1     | 15  | 14  |
|     |     |      GND |      |   |  9 || 10 | 1 | ALT5 | RXD1     | 16  | 15  |
|  17 |   0 |   GPIO17 |   IN | 1 | 11 || 12 | 1 | IN   | GPIO18   | 1   | 18  |
|  27 |   2 |   GPIO27 |   IN | 1 | 13 || 14 |   |      | GND      |     |     |
|  22 |   3 |   GPIO22 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO23   | 4   | 23  |
|     |     |     3.3V |      |   | 17 || 18 | 0 | IN   | GPIO24   | 5   | 24  |
|  10 |  12 | SPI0MOSI | ALT0 | 0 | 19 || 20 |   |      | GND      |     |     |
|   9 |  13 | SPI0MISO | ALT0 | 0 | 21 || 22 | 0 | IN   | GPIO25   | 6   | 25  |
|  11 |  14 | SPI0SCLK | ALT0 | 0 | 23 || 24 | 1 | OUT  | GPIO8    | 10  | 8   |
|     |     |      GND |      |   | 25 || 26 | 1 | OUT  | GPIO7    | 11  | 7   |
|   0 |  30 |    GPIO0 |   IN | 1 | 27 || 28 | 1 | IN   | GPIO1    | 31  | 1   |
|   5 |  21 |    GPIO5 |   IN | 1 | 29 || 30 |   |      | GND      |     |     |
|   6 |  22 |    GPIO6 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO12   | 26  | 12  |
|  13 |  23 |   GPIO13 |   IN | 0 | 33 || 34 |   |      | GND      |     |     |
|  19 |  24 |   GPIO19 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO16   | 27  | 16  |
|  26 |  25 |   GPIO26 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO20   | 28  | 20  |
|     |     |      GND |      |   | 39 || 40 | 0 | IN   | GPIO21   | 29  | 21  |
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
| sOc | iNo |   Name   | Mode | V | Ph || Ph | V | Mode |   Name   | iNo | sOc |
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
```