epsilonrt/piduino #34: Raspberry Pi OS bullseye build

Eduardo-bat, 2024-02-13:
The Raspbian section of the main page says

> The Raspbian repository provides Piduino packages for armhf architecture for Stretch and Buster only.

This means that piduino can't work on Raspberry Pi OS bullseye? I've tried installing the debian bullseye package, raspbian buster package and building from source. The result is always the same: a successful installation, `pido` returns
`
pido: Command expected !
pido -h or man pido for help.
`
`pinfo` returns
`
Name            : RaspberryPi 2B
Family          : RaspberryPi
Database Id     : 21
Manufacturer    : Embest
Board Revision  : 0xa21041
SoC             : Bcm2709 (Broadcom)
Memory          : 1024MB
GPIO Id         : 3
PCB Revision    : 1.1
I2C Buses       : /dev/i2c-2
`
but `pido readall`, `sudo pido readall` and `pido readall 1` returns
`pido: invalid argument (Unable to read pull resistor) !`

Is there a way to use piduino and its packages on Raspberry Pi OS bullseye? If so, how?

epsilonrt, 2025-04-09:
I will release a new debian package version for RaspiOS buster, bullseye and bookworn before the end of the day (armhf and arm64 architecture).

epsilonrt, 2025-04-09:
```
epsilonrt@rpi-bullseye-armhf:~/src $ pinfo
Name            : RaspberryPi 2B
Family          : RaspberryPi
Database Id     : 20
Manufacturer    : Sony UK
Board Revision  : 0xa01041
SoC             : Bcm2709 (Broadcom)
Memory          : 1024MB
GPIO Id         : 3
PCB Revision    : 1.1
I2C Buses       : /dev/i2c-1,/dev/i2c-2
SPI Buses       : /dev/spidev0.0,/dev/spidev0.1
Serial Ports    : /dev/ttyAMA0
epsilonrt@rpi-bullseye-armhf:~/src $ pinfo -v
Piduino pinfo version 0.5-42
Copyright © 2018-2019 Pascal JEAN, https://github.com/epsilonrt/piduino
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'pinfo -w' for details.

epsilonrt@rpi-bullseye-armhf:~/src $ pido readall
                                    J8 (#1)
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
| sOc | iNo |   Name   | Mode | V | Ph || Ph | V | Mode |   Name   | iNo | sOc |
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
|     |     |     3.3V |      |   |  1 || 2  |   |      | 5V       |     |     |
|   2 |   8 |     SDA1 | ALT0 | 1 |  3 || 4  |   |      | 5V       |     |     |
|   3 |   9 |     SCL1 | ALT0 | 1 |  5 || 6  |   |      | GND      |     |     |
|   4 |   7 |    GPIO4 |   IN | 1 |  7 || 8  | 1 | ALT0 | TXD0     | 15  | 14  |
|     |     |      GND |      |   |  9 || 10 | 1 | ALT0 | RXD0     | 16  | 15  |
|  17 |   0 |   GPIO17 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO18   | 1   | 18  |
|  27 |   2 |   GPIO27 |   IN | 0 | 13 || 14 |   |      | GND      |     |     |
|  22 |   3 |   GPIO22 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO23   | 4   | 23  |
|     |     |     3.3V |      |   | 17 || 18 | 0 | IN   | GPIO24   | 5   | 24  |
|  10 |  12 | SPI0MOSI | ALT0 | 0 | 19 || 20 |   |      | GND      |     |     |
|   9 |  13 | SPI0MISO | ALT0 | 0 | 21 || 22 | 0 | IN   | GPIO25   | 6   | 25  |
|  11 |  14 | SPI0SCLK | ALT0 | 0 | 23 || 24 | 1 | OUT  | GPIO8    | 10  | 8   |
|     |     |      GND |      |   | 25 || 26 | 1 | OUT  | GPIO7    | 11  | 7   |
|   0 |  30 |    GPIO0 |   IN | 1 | 27 || 28 | 1 | IN   | GPIO1    | 31  | 1   |
|   5 |  21 |    GPIO5 |   IN | 1 | 29 || 30 |   |      | GND      |     |     |
|   6 |  22 |    GPIO6 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO12   | 26  | 12  |
|  13 |  23 |   GPIO13 |   IN | 0 | 33 || 34 |   |      | GND      |     |     |
|  19 |  24 |   GPIO19 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO16   | 27  | 16  |
|  26 |  25 |   GPIO26 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO20   | 28  | 20  |
|     |     |      GND |      |   | 39 || 40 | 0 | IN   | GPIO21   | 29  | 21  |
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
| sOc | iNo |   Name   | Mode | V | Ph || Ph | V | Mode |   Name   | iNo | sOc |
+-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
```